cp2004/OctoPrint-WS281x_LED_Status #171: [Request]:

mistfink, 2021-12-02:
### What would you like implemented?

Helloo, first of all, very nice plugin, i love it!
But
It would be great if there is an option to change the GPIO-"LED Data in"-pin to another pin (like to gpio 20)
i dont know if it is important that it has to be an SPIO MOSI-pin or any other GPIO.

### Why should this be implemented?

I have installed the touchscreen and the GPIO10 is used by that. But i want to use the touchscreen AND the LED status lights
(and a power button but that's an other thing :)  )

cp2004, 2021-12-02:
You can change the GPIO pin in the LED settings. It does have to be an SPI capable pin, and you may have to configure some extra flags in the config.txt file to enable the second SPI interface. GPIO 20 is another SPI pin: https://pinout.xyz/pinout/pin38_gpio20

cp2004, 2021-12-02:
Also note that only GPIO 10 (SPI0) is available all Pis, it can get confusing with all the different hardware - https://www.raspberrypi.com/documentation/computers/raspberry-pi.html#spi-hardware

closed 2021-12-27 by github-actions[bot]