cp2004/OctoPrint-WS281x_LED_Status #174: Lights not working

RckyBooby, 2022-01-02:
### Describe your problem

I have had these lights set up for about a week not and they have just stopped working. they were working great for a few days with all of the effects I had programmed working correctly but I had turned off the system completely because I was leaving for a couple of days and I did not want to just let my printer and raspberry pi just hang out on and not doing anything. I came home and now they won't turn on, everything else functions fine, and in the OctoPrint web interface the plugin runs fine as I can toggle on torch mode and turn the lights on and off even though the lights themselves don't turn on. I have ran the OS Config Test and it passes everything.

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7799894/plugin_ws281x_led_status_debug.log)
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7799895/octoprint.log)

cp2004, 2022-01-02:
Presumably you checked all the hardware such as the power supply etc. are still working? All the wiring is still good and nothing got bumped or unplugged?

RckyBooby, 2022-01-03:
That is correct, I have checked all of the connections and everything is tight. I just canceled a print for an unrelated issue and the first three lights have come on

closed 2022-01-25 by github-actions[bot]