cp2004/OctoPrint-WS281x_LED_Status #175: Config is Correct but LEDs Don't Turn On

Galador81, 2022-01-09:
### Describe your problem

I have been trying to get my LEDs working for a while now but no dice. I'm very new to all of this stuff too so keep that in mind. 

Nothing is lighting up on them. I was trying to use a different plugin originally. That one would light up a few randomly and with the wrong colors. Two of the LED's wouldnt even work while the others were. I am using a Adafruit NeoPixel Jewel 7 (RGBW - Cool white). I am also using a Pi Zero 2 w for this. 

My wiring is correct. I've been occasionally testing it on an Arduino, and they work like a charm. I know the Pi outputs enough power for the LED's since I left it wired to the Pi and plugged in the Data pin to the Arduino.

I have the 5v and Ground wired onto the Pi and for the data pin I have been using pin 12 (GPIO 18). I am fairly certain they are SK6812s too. The Config tool worked for everything and it's all showing green checks marks. I'm really not sure where to go from here so any ideas would be much appreciated.

Let me know if I can provide any more info. Thanks.

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint-logs.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7834796/octoprint-logs.zip)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7834798/plugin_ws281x_led_status_debug.log)

cp2004, 2022-01-09:
> I have been using pin 12 (GPIO 18). I changed the data pin to 12 (from 10) in the settings as well.

This is your problem. GPIO 18 is not a supported pin for controlling the LEDs through SPI.

Read the 'GPIO pin options' section here https://cp2004.gitbook.io/ws281x-led-status/configuration/led-strip-configuration

Galador81, 2022-01-09:
> This is your problem. GPIO 18 is not a supported pin for controlling the LEDs through SPI.
>
> Read the 'GPIO pin options' section here [...]

Thanks. I remember seeing this and not thinking much of it. Can't believe I missed that. It's working now.

cp2004, 2022-01-09:
👍

closed 2022-01-09 by cp2004