cp2004/OctoPrint-WS281x_LED_Status #177: Question

jtannenba, 2022-01-20:
### Describe your problem

Is there a way to tell the led to shut off when the print is done? Did I miss a setting?  I only use startup and print.  When the Psu control shuts the printer down the light is still on.

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

cp2004, 2022-01-20:
If you are using PSU control, then you can add `@WS_LIGHTSOFF` to the gcode scripts to also control the LEDs.

There is an option in this plugin to have the LEDs go off after a certain time, though it is a bit more complicated. You need the 'Success' and 'Idle' effects enabled, and then a number of seconds for 'Return to idle' and 'idle timeout'.

jtannenba, 2022-01-20:
Thank you.  What I did was run the external 5 v from the neopixels to the plug controlled by the PSU.  I don't know why dum dum didn't think of that before. :-)

closed 2022-01-20 by jtannenba