cp2004/OctoPrint-WS281x_LED_Status #178: OctoDash - UCTronics UC-640 - WS281x Status (SPI?) compatibility

TyropeDeMotorChope, 2022-01-23:
### Describe your problem

I used [this](https://github.com/UnchartedBull/OctoDash/wiki/User-Guides#uctronics-lcd35-spi-install) guide to install OctoDash on a raspberry pi 4b, and had it running. I installed ws281x led status and went through the configs, and now screen doesn't work and I can't even connect to the Pi over SSH... that is, unless I take the screen off the pin headers. If I have the screen plugged in currently, the pi is a brick with a backlight.

Because the screen I am using takes over the gpio 10 pin, I am forced to use gpio 20. I have added what I think I should in the config file, which I've attached.

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[config.txt](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7921338/config.txt)

[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7921356/octoprint.log)

[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7921357/plugin_ws281x_led_status_debug.log)

cp2004, 2022-01-23:
What GPIO pin have you set in the settings?

cp2004, 2022-01-23:
When I try and set GPIO 20, I get a different error to the one you are seeing, about the GPIO number not being allowed.

TyropeDeMotorChope, 2022-01-24:
I definitely didn't even get to changing it there before I ran into problems. I have now, and the LEDs and Screen still both don't work.
[config.txt](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7922130/config.txt)
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7922131/octoprint.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/7922132/plugin_ws281x_led_status_debug.log)

Here's the files after changing it to 20 in the plugin settings.

TyropeDeMotorChope, 2022-01-24:
I found through [this](https://github.com/jgarff/rpi_ws281x), that you can't use two SPI devices at once and the screen is SPI. Do you know if there is a PWM or PCM solution to NeoPixels with either OctoPrint or a MoonRaker setup?

cp2004, 2022-01-24:
You can use PWM (GPIO 18) through this plugin, but it requires running as root, which is not recommended for a web server generally, and requires the `--iknowwhatimdoing` flag to acknowledge the security risk when you run OctoPrint.

cp2004, 2022-02-04:
Hopefully all your questions were answered - if there's anything else you want to know about the plugin, please ask away! I'll close this for now but it can always be reopened.

closed 2022-02-04 by cp2004