cp2004/OctoPrint-WS281x_LED_Status #180: [Bug]: Auto activate torch while viewing webcam does not work when using OctoEverywhere! or OctoApp

Scott-28, 2022-02-09:
### Description of the bug

I have the "Torch Mode is a toggle" and "Auto activate torch when viewing webcam stream" check boxes both checked which work great to turn on torch mode while using OctoPrint from my PCs web browser but when viewing the webcam via OctoEverywhere! or OctoApp plugin on my phone, torch mode does not turn on. 

### Steps to reproduce

1. Install OctoEverywhere! plugin.
2. Sign-in to OctoEverywhere! account on mobile phone browser or external screen hooked to RaspberryPi.
3. Click the webcam/picture button to view webcam, torch mode will not turn on.

Similar process for OctoApp but OctoApp on the main page displays a live feed of the camera so I imagine getting torch mode to function while using OctoApp will be a different method/set of problems than using OctoEverywhere!

### What happened instead?

Torch mode does not turn on and the light configuration stays on my "idle" settings.

### Plugin version

0.8.0

### OctoPrint version

1.7.3

### (if relevant) OctoPi version

0.18.0

cp2004, 2022-02-09:
Auto on torch can only work when the webcam stream is viewed in the OctoPrint UI. Yes that's quite limited but since OctoPrint doesn't actually provide the webcam, I have no way of knowing if it is being used externally in an app.

Last time I checked, the OctoApp author had implemented this feature into their app. You may have to enable this in the app's settings:

Scott-28, 2022-02-09:
Thank you! I didn't realize the app had that option. 
I understand, thanks for the info, I figured it wasn't going to be an easy (or possible) thing to accomplish or it would have already been done.

closed 2022-02-09 by Scott-28