cp2004/OctoPrint-WS281x_LED_Status #181: Process event dose not start

HerrHaseGermany, 2022-02-09:
### Describe your problem

After warm up the led stripe stays red. it seems that it switches to the "Sets LEDs to a custom effect until the next event happens." and stays there... i dont know what to do

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8034355/octoprint.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8034365/plugin_ws281x_led_status_debug.log)

HerrHaseGermany, 2022-02-09:
this happend after update to 0.8 before was fine

HerrHaseGermany, 2022-02-09:
rechecked, after changing "Sets LEDs to a custom effect until the next event happens." it happens the same way, staying red after warmup

cp2004, 2022-02-10:
Thanks for the bug report. I'm away at the moment but will take a look as soon as I can.

HerrHaseGermany, 2022-02-21:
i have observed some prints now, what i can say is, that the process event will start eventualy, but sometimes long after the print has started

cp2004, 2022-03-01:
Confirmed there is a bug here, needs fixing.

cp2004, 2022-03-01:
Fixed, will be released as 0.8.1 when I get the other open bug report fixed as well.

cp2004, 2022-03-13:
0.8.1 has been released.

closed 2022-03-13 by cp2004