cp2004/OctoPrint-WS281x_LED_Status #183: [Bug]: LEDs are in Heat mode after printing has started

JonSnow88, 2022-02-27:
### Description of the bug

I will start a print and it will start to heat as normal and LEDs will light Blue to Red, however when the print actually starts the LEDs stay red, until half the first layer has printed then it switches to monitoring the print progress.

I also have it set so it only tracks the bed heating and not the hotend, but that seems to be ignored. This was so that once the bed is heated the LEDS do not reset for hotend heating and LEDs just stay red until printing starts.

Start Bed Heating > LEDs Blue to Red > Start Hotend Heating > LEDs reset and start the Blue to Red sequence again.

### Steps to reproduce

start print
leds start changing from blue to red
print starts and leds are still red
first layer starts
leds are red
halfway in first layer leds changes to print progress 

### What happened instead?

This was not happing on the old version, only started when i upgraded. 

### Plugin version

latest

### OctoPrint version

1.7.3

cp2004, 2022-03-01:
So are there two issues here?

1. Disabling tracking tool heating is broken
2. After heating, progress bar is not replaced?

Unfortunately I can't look into them without the log file. The `octoprint.log` and `plugin_ws281x_led_status_debug.log` preferably with 'Debug logging' checked on the features tab.

cp2004, 2022-03-13:
Any news here @JonSnow88?

biancapascal, 2022-03-18:
The problem with the led's are still not solved. 
The leds are still jumping in a strange way in the color white.
If i load version 0.7 everything works normal, but in previous version and this 0.8 releas its not.
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8304789/plugin_ws281x_led_status_debug.log)
.

cp2004, 2022-03-18:
@biancapascal that is not related to this issue. If you have a different issue please open your own bug report so the issues do not conflict.

cp2004, 2022-03-27:
@JonSnow88 I need replies to my questions if I'm going to investigate bug reports please. Nearly a month on from opening the issue.

closed 2022-03-27 by cp2004