cp2004/OctoPrint-WS281x_LED_Status #186: Using GPIO18 for LED strip instead of GPIO10

gargamel778, 2022-03-12:
### Describe your problem

Hi,

The plugin works as expected when connected to GPIO10. However, I have an ADXL345 accelerometer (for Klipper input shaping resonances measurements) connected over SPI and taking up GPIO10. I could not make them share the GPIO; when the LED strip is connected to GPIO10 in parallel - Klipper rejects the accelerometer when running ACCELEROMETER_QUERY.

I want to use GPIO18 (or GPIO21) for my Neopixel strip. When configuring the plugin this way, the plugin log shows 

> ERROR: RuntimeError('ws2811_init failed with code -5 (mmap() failed)')

I tried various ways to overcome -

- add the user pi to the gpio group
- added 'dtoverlay=spi1-3cs' to /boot/config.txt

I have tried running strandtest.py from ~/rpi-ws281x-python/examples as regular user and it fails with the following (similar) error -

> Can't open /dev/mem: Permission denied
Traceback (most recent call last):
  File "strandtest.py", line 97, in <module>
    strip.begin()
  File "/home/pi/.local/lib/python3.7/site-packages/rpi_ws281x/rpi_ws281x.py", line 131, in begin
    raise RuntimeError('ws2811_init failed with code {0} ({1})'.format(resp, str_resp))
RuntimeError: ws2811_init failed with code -5 (mmap() failed)
Segmentation fault

Running strandtest.py as root with sudo works correctly. I'm reluctant to run Octoprint as root user, given the security issues involved with it.

Will appreciate any help.

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8237501/octoprint.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8237502/plugin_ws281x_led_status_debug.log)

gargamel778, 2022-03-12:
I also tried GPIO20, following "What other pins work?" in https://cp2004.gitbook.io/ws281x-led-status/configuration/led-strip-configuration#gpio-pin-options - the result is the ~/.octoprint/logs/plugin_ws281x_led_status_debug.log shows

> ERROR: RuntimeError('ws2811_init failed with code -11 (Selected GPIO not possible)')

cp2004, 2022-03-12:
I need to update that doc actually - there's an issue with the library that means that GPIO 20 doesn't work either. I tested it, and then looked at the code - it limits itself to GPIO 10 for SPI only. I haven't had a chance to try editing the rpi-ws281x library yet to see if I could add support - as far as I understand the system, it should work the same, but it is currently blocked.

Is it possible to change the pin for the Klipper sensor? That might be easier to do than trying to make this work, but I have no idea what's possible there so it may also be harder...

The PWM pins (18) require root access, and I don't know any way around this.

gargamel778, 2022-03-12:
Thanks for your swift response! 
I don't think there's any other way to wire the accelerometer, according to [this document](https://www.klipper3d.org/Measuring_Resonances.html).

If that's the case, I will wire the LED strip to the printer's mainboard (BTT SKR 2) dedicated neopixel port and control the LEDs from Klipper. However, this way I will have less fancy LED control effects.

cp2004, 2022-03-12:
Probably both this plugin and Klipper are only created to use SPI0, not expandable to other SPI options available on newer Pis.

It *may* work if this section was edited: https://github.com/jgarff/rpi_ws281x/blob/master/ws2811.c#L679-L681 but maybe something else will break further down the line. I'm not really sure, and I don't really understand the code in there at all, with no time to learn it...

I've previously looked into writing my own driver for supporting *just* SPI (making it significantly simpler) but I would probably have to learn C/C++ or something first...

gargamel778, 2022-03-12:
When connecting the LED strip directly to the 3D printer's mainboard and controlling it with Klipper firmware I will skip this issue altogether. This isn't the route I preferred but it'll do.

Thanks anyway!

closed 2022-03-12 by cp2004