cp2004/OctoPrint-WS281x_LED_Status #188: Unable to manually disable/turn off 'torch' after manually enabling.

FreeManFreeThinker, 2022-04-04:
### Describe your problem

No matter how I have the torch button settings configured (toggle, timed, color, etc...), I am unable to manually disable/turn off my lights after having manually enabled them with the torch button. I have to go into the plugin settings and custom test the color black to get them to turn off. 

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8405969/octoprint.log)

[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8405971/plugin_ws281x_led_status_debug.log)

cp2004, 2022-04-04:
If you enable the idle feature, does it work correctly? There's probably some issues with certain combinations of settings I've not tried.

FreeManFreeThinker, 2022-04-05:
> If you enable the idle feature, does it work correctly? There's probably some issues with certain combinations of settings I've not tried.

I furthered my troubleshooting extensively and eventually discovered a faulty raspberry pi PSU to be the root of all the problems I was experiencing. It's safe to say that was the problem seeing that the issue cleared up after replacing the power supply. Solved (user error, haha)!

closed 2022-04-05 by FreeManFreeThinker

cp2004, 2022-04-05:
Well that's good to hear, as it means I don't have to go bug-hunting... 🙂