cp2004/OctoPrint-WS281x_LED_Status #189: ERROR: RuntimeError('ws2811_init failed with code -13 (Unable to initialize SPI)')

volt64bolt, 2022-04-14:
### Describe your problem

I can't get the lights to turn on, at first I thoughts the lights were dodgy but I checked the logs and found the line `ERROR: RuntimeError('ws2811_init failed with code -13 (Unable to initialize SPI)')` (they might still be dodgy lights but this seems like the main culprit)
this log is without debug logging enabled:
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8491459/plugin_ws281x_led_status_debug.log)
 this is after i enabled debug logging:
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8491466/plugin_ws281x_led_status_debug.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8491472/plugin_ws281x_led_status_debug.log)

I am running a raspberry pi 4b and ws2812 5v leds, powered with a 600ma 4.5v power supply, can change this from 3v to 5v and have tested on each setting, attached below are images of the setup, with the pins unplugged. I did triple check I plugged them into the right slots
[image]
[image]
[image]
[image]
 

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8491473/plugin_ws281x_led_status_debug.log)
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8491474/octoprint.log)

volt64bolt, 2022-04-15:
how do i close this now?

cp2004, 2022-04-15:
There should usually be a close button at the bottom, but I can do it too.

closed 2022-04-15 by cp2004