cp2004/OctoPrint-WS281x_LED_Status #194: [Request]: Add support for SPI1 LEDs

cp2004, 2022-06-15:
### What would you like implemented?

SPI1 is not able to be used, only SPI0 (GPIO 10)

The library rpi_ws281x limits this. Try and patch this upstream, then bump the dependency in the plugin

### Why should this be implemented?

Would allow for slightly more freedom with what GPIO can be used (or for multiple programs to use SPI)

cp2004, 2023-01-10:
Try this:

Add the following additional lines to `/boot/config.txt`:
```
dtoverlay=spi1-1cs
```
Install the patched up dependency:

[rpi_ws281x-4.3.4.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/10383979/rpi_ws281x-4.3.4.zip)

Either download & then use plugin manager 'from file', or enter the below URL in 'from URL':

```
https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/10383979/rpi_ws281x-4.3.4.zip
```

Unlikely to work, but you never know

patricknitsch, 2025-05-11:
Does this dependency still works? I configured it and use the actual 0.8.1 Does it also works there? At the moment with pin 20 the led's doesn't work.