cp2004/OctoPrint-WS281x_LED_Status #196: LED #1 Not Changing

microshot71, 2022-06-22:
### Description of the bug

I had trouble getting my 60 LED WS2812 to work; all 60 LEDs would all change to the same random color each time I attempted to choose a specific color in the utilities tab. I added a 470 ohm resistor and that corrected the problem. 59 LEDs now respond as expected. However, the first LED is "stuck" on green. I double checked to make sure I unchecked the "Skip first LED" box in the LED strip editor; it is unchecked. Any suggestions?

Great product, by the way.

Thanks.

### Steps to reproduce

1. Choose test Red, Green, or Blue. 
2. Observed LEDs.
3. All but the first LED changed color to expected color.

### What happened instead?

First LED on the string of 60 does not change color as expected.

### Plugin version

 Version 0.8.1  

### OctoPrint version

 OctoPrint 1.8.1 

### (if relevant) OctoPi version

OctoPi 0.18.0

### Log Files & Screenshots

[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8960136/plugin_ws281x_led_status_debug.log)
[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/8960139/octoprint.log)

### Anything else?

n/a

cp2004, 2022-06-24:
Apologies for the delay replying here, I haven't had time to properly look at this issue and wonder what is going on.

How have you setup the wiring for the LEDs here? Is it using a level shifter or just going straight with the 3v3 signal from the Pi?

The situation you are in sounds very similar to the use case for the 'Skip first LED' where people would use one LED to sort out the signal so it was acceptable for all the LEDs. Since the way WS281x LEDs work is by passing the signal on from one LED to the next sequentially, once it is through the first one it works better for the others. If you get what I mean.

So I'm wondering if maybe the signal is not 'good enough' for the LEDs. I have never actually seen this kind of problem myself, so I generally try and make up some theories to help. The main question is how have you wired everything up.

cp2004, 2022-07-26:
@microshot71 do you still need help with this? Or is the problem now fixed and I can close the issue?

microshot71, 2022-07-27:
Yeah, could use help. Still can’t get the first LED to change. I’ve thought about adding a single addressable LED before the string of lights and have it be located next to the Raspberrypi case to be an indicator that the power is on. Any ideas welcome.

Thanks

Greg

cp2004, 2022-07-27:
Please answer the question about wiring I put above a month ago, so I can understand your setup or I can't help.

microshot71, 2022-07-27:
Sorry, didn’t get an email notification about the question above. Yes, I am using a EPLZON TXS0108E High Speed Full Duplex Logic Level shifter. I think I’m going to add a single addressable LED before the LED strip but after the shifter and resistor, to “sort out the signal”. If that doesn’t work, I’ll post another ticket.

cp2004, 2022-07-28:
I haven't used that level shifter, so I don't know if it is a good one or not. Have you tried it without the level shifter to see if it just works with no signal conversion? That works for *most* LEDs, provided the signal wire is not long to introduce more voltage drop.

cp2004, 2022-08-30:
@microshot71 *gentle ping* if you still need help...

cp2004, 2022-09-17:
Since there has been no reply, I will close the issue assuming the problem is solved. If you still need help you can comment below still.

closed 2022-09-17 by cp2004