cp2004/OctoPrint-WS281x_LED_Status #198: connection to printer

sourekv, 2022-08-30:
### Describe your problem

After setting up plugin are leds working but I am unable co connect to printer. When I disable the plugin connection works. Where should I start looking? Thank you.

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint-logs.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/9454677/octoprint-logs.zip)
[octoprint-systeminfo-20220830181116.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/9454678/octoprint-systeminfo-20220830181116.zip)

sourekv, 2022-08-30:
[plugin_ws281x_led_status_debug.zip](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/9454917/plugin_ws281x_led_status_debug.zip)

cp2004, 2022-08-30:
Can you try unchecking the 'Request Exclusive access' option in the serial settings?

[image]

There is this long-standing issue with certain printers, and over the last 2 and a bit years I have maintained this plugin I've had 3 (now 4) reports of this issue. You can see one of them here where I also describe another workaround - #113.

I'm not sure how to fix it, as I have not yet found anything I can do differently with the effect process that will stop it inheriting the serial port as well. I can't reproduce the issue to work out exactly what is going on, these are just suspicions.

sourekv, 2022-08-30:
I will try that and report back. So far is the issue intermitent.

sourekv, 2022-09-05:
Thanks for your advice. Your suggestion - unchecking exclusive access has solved the issue.

closed 2022-09-05 by sourekv

PetrP2B, 2023-01-19:
Hi, I have the same problem, plugin WS2812 and Prusa i3 MK3S+. After uncheck "Request exclusive access to the serial port" the problem does not occur.