cp2004/OctoPrint-WS281x_LED_Status #199: Plugin doesn't change to printing affect for several minutes after printing starts.

Tr33x0rs, 2022-09-20:
### Describe your problem

I recently upgraded my Octoprint version on a new SD card. I had been previously using this plugin and it worked great. Everything is connected the same and basically everything works. One issue I'm seeing is that once the tool and bed heating is over the printer will start printing and my lights all stay red (in the heating effect) for 5-10 minutes. I looked at all my previous settings and everything matches. I looked at my logs and see where it changed to the printing effect but I'm not sure why it happened so long after printing began. Thank you in advance. 

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint (13).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/9604836/octoprint.13.log)
[plugin_ws281x_led_status_debug (10).log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/9604837/plugin_ws281x_led_status_debug.10.log)

cp2004, 2022-10-05:
Apologies, I haven't had a chance to look at this yet. I will try and get to it at some point soon.

cp2004, 2022-10-13:
So I finally got a chance to properly look into the logs here (sorry about that delay...).

From what I can see, the plugin is only switching to the print progress effect when it reaches 1% complete on the print - for a short print, this is very soon, for a long print, this is likely to be much later on. It has been a while since I have worked on it (more hopefully coming soon once I get on top of some stuff), but this seems like it maybe missed the opportunity to remember the progress needed to be shown.

In my opinion the whole code (& concept) for switching states and back again when stuff is finished, and what effects take priority over each other is a bit too complicated still and it is on my list to look at again with a different method. It has numerous bugs depending on the configuration of the plugin and how your printer works, which is not great. All I think I can say is that I hope to work on it to make this kind of problem less common.

TLDR; I am not sure why it didn't switch states, and the state code is not very reliable or easy to handle in my opinion...