cp2004/OctoPrint-WS281x_LED_Status #202: [Bug]: Neopixel Ring not working correctly

cxen, 2022-11-06:
### Description of the bug

Device info: NeoPixel Ring - 16 x 5050 RGBW LEDs w/ Integrated Drivers - Cool White https://www.adafruit.com/product/2856

I have the above ring LED light connected directly to a Pi 4B, GPIO pin 10. WS281x_LED_Status can only use it correctly only if I use SK6812_STRIP_GRBW, when it should be WS2812_STRIP that is also GRBW, which is not available. Most other Strip Types do not work, only work for RBG or the color buttons light the wrong color LEDs.

The main issues is that the dashboard/top bar button does not work after every reboot, unless I go in the WS281x_LED_Status and press one of the Test light buttons.

### What happened instead?

The dashboard bar button does not work after reboots.

### Plugin version

0.8.1

### OctoPrint version

1.8.6

### (if relevant) OctoPi version

Version 0.18.0, running on Raspberry Pi 4 Model B Rev 1.4

cp2004, 2022-11-06:
> Log Files & Screenshots
> No response

Please, upload the `plugin_ws281x_led_status_debug.log` file, preferably with debug logging enabled, as this is a very helpful thing to have...

cxen, 2022-11-07:
> > Log Files & Screenshots
> > No response
> 
> Please, upload the `plugin_ws281x_led_status_debug.log` file, preferably with debug logging enabled, as this is a very helpful thing to have...

[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/9953846/plugin_ws281x_led_status_debug.log)

I enabled debug logging at [2022-11-07 17:53:09,982]

cxen, 2022-12-06:
Any update on this issue?

cp2004, 2023-01-02:
Really sorry this issue was missed - I have had a couple that went missing that it seems I wasn't notified about.

The good news is that I can see the issue exactly as you described from reading the log files, the plugin is switching the lights on/off but there is no effect running at that moment point in time so nothing is being displayed by the looks of it.

```
[2022-11-07 17:55:25,687] DEBUG: New message: {'type': 'lights', 'action': 'on'}
[2022-11-07 17:55:25,688] INFO: Switching lights on
[2022-11-07 17:55:25,690] DEBUG: Changing effect to blank
[2022-11-07 17:55:25,690] DEBUG: Blanking LEDs
[2022-11-07 17:55:29,412] DEBUG: New message: {'type': 'lights', 'action': 'off'}
[2022-11-07 17:55:29,412] INFO: Switching lights off
[2022-11-07 17:55:29,915] DEBUG: Changing effect to blank
[2022-11-07 17:55:29,916] DEBUG: Blanking LEDs
[2022-11-07 17:55:31,623] DEBUG: New message: {'type': 'lights', 'action': 'on'}
[2022-11-07 17:55:31,623] INFO: Switching lights on
[2022-11-07 17:55:31,625] DEBUG: Changing effect to blank
[2022-11-07 17:55:31,625] DEBUG: Blanking LEDs
[2022-11-07 17:55:40,770] DEBUG: New message: {'effect': 'Solid Color', 'color': '#ffffff', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
``

This is because you have all the effects disabled except for the 'Pause' effect. There is nothing to run on the script, so it doesn't do anything. Once the printer has paused and this effect is triggered, I would expect that the on/off toggle would work effectively. Does that make sense?

There is a high chance that there could be some actual bugs when disabling a lot of effects, as the code logic that processes this is not that good. I intend to rewrite it at some point to be more robust - issues #204, #199, #195 are all related to this idea.

Again, sorry for the *really really* long delay in replying, I was very busy in November and then didn't manage to catch up on the notifications properly. I hope to be working on this plugin to add segment control & fix some of the bugs in the near future.