cp2004/OctoPrint-WS281x_LED_Status #203: [Request]: A way to control the lights from Homeassistant

sendorm, 2022-11-22:
### What would you like implemented?

The current octoprint integration of homeassistant using mqtt supports many features. Such as, starting stopping pausing prints, making a connection, turning printer power on/off, getting the ETA, print duration, sensor temperatures etc. There is also a great card which creates a unique user interface for octoprint. One thing missing is the ability to turn on/off the connected leds.

### Why should this be implemented?

It would be great to be able to turn on/off the leds connected to octoprint from homeassistant. And if using the mqtt integration the leds will just appear on homeassistant.

cp2004, 2022-11-26:
I'm not sure what I would need to do to support this, and I don't really want to add specific things like this to the plugin - it's a lot for me to maintain for something I wouldn't use. If the home assistant integration allows you to send gcode commands to the printer, take a look at the @ commands that can be sent like gcode to control the LEDs.

https://cp2004.gitbook.io/ws281x-led-status/documentation/host-commands

sendorm, 2022-11-26:
I think it would be better to ask the homeassistant integration developers for this. Thanks for leading me to the right direction. They could just make octoprint-wled appear as a light device on homeassistant.

closed 2022-11-26 by sendorm