cp2004/OctoPrint-WS281x_LED_Status #204: [Bug]: Torch does not turn off

JumpmanJunior, 2022-11-23:
### Description of the bug

I want to use only torch mode. The LEDs turn on if I e.g. press the torch button or if I configure it to turn on automatically when viewing the webcam. But they do not turn off after the configured time frame or after pressing the button again if enabled toggle. They do turn off if I e.g. save the settings or do this via the Utilities page.

### Steps to reproduce

1. Enable torch mode and configure that it lasts for e.g. 5 seconds
2. Save settings. LEDs are off.
3. Press torch button at the top. LEDs turn on.
4. Wait for 5 seconds. LEDs stay on.

### What happened instead?

LEDs stay on.

### Plugin version

0.8.1

### OctoPrint version

1.8.6

### (if relevant) OctoPi version

0.18.0

cp2004, 2022-11-26:
> ### Log Files & Screenshots
> *No response*

Please enable debug logging, and then upload the `plugin_ws281x_led_status_debug.log` file after you have reproduced the issue.

JumpmanJunior, 2022-11-27:
[debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/10097461/debug.log)

You can see in the log that I first try it with a timeout of 4 seconds and then with enabled toggle. In both cases the torch does not turn off. When I press the torch button to disable it nothing gets logged.

cp2004, 2022-11-29:
This is likely a bug that has been triggered by all the other effects being disabled. The way the plugin turns the torch off is by replacing it with whatever effect was running previously. Here that gets blocked because everything is disabled.

As a workaround, you could try enabling the startup effect, set it to solid colour black (so it's like it's off) and then if I understand correctly the torch will work as intended.

This part of the code is quite fragile as there are loads of different conditions to work through, at some point I plan to rewrite how it works but it won't be for a while - hopefully the workaround works as I won't be able to fix the issue at the moment.

kevinvalk, 2023-02-07:
For the workaround to work, I had to reboot the raspberry when settings the off color on startup effect before the torch worked correctly.