cp2004/OctoPrint-WS281x_LED_Status #205: impossible to control my ws2812 strip

djepsylon, 2022-11-25:
### Describe your problem

i try to control 29 leds on a ws2812 strip on the gpio18 pin.
before i re install all my octoprint setup it work correctly but now nothing work i don't undrstnd why.
i have a ws2812 strip on his power supply, gpio 18 pin on the signal pin of the strip, +5v of the power supply on the +5v pin of the strip and the gnd of the poxer supply and the gnd of the rapberry on the gnd pin of the print. i've tryed with other gpio pin and it don't work too. i think my gpio is ok beacause ai have 2 runout sensors on the two octprint's instances who control 2 different printers and it work correcly.
nothing have changed in the wirinbg and i've controled all and all is ok.
i think the problem is in the software but i don't have less knowlege for investigate more.
sorry for my bad english, it's not my native language.
[image]

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/10095201/octoprint.log)
[plugin_ws281x_led_status_debug.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/10095203/plugin_ws281x_led_status_debug.log)

cp2004, 2022-11-26:
> i try to control 29 leds on a ws2812 strip on the gpio18 pin.

Please see the documentation on how to wire your LEDs, about GPIO18. Usually people will use GPIO10 with this plugin.

https://cp2004.gitbook.io/ws281x-led-status/configuration/led-strip-configuration#what-other-pins-work

To use a PWM pin, you must configure OctoPrint to run as root, which is not recommended for security.

djepsylon, 2022-11-26:
hi,
thanks for your answer.
i've tried with gpio10 too and nothing works too

cp2004, 2022-11-29:
Could you upload a new log (`plugin_ws281x_led_status_debug.log`) since it has been switched in the config to GPIO 10? So I can see if the plugin thinks it is working OK.

cp2004, 2023-01-02:
@djepsylon If you still need help please do reply.

imkebe, 2023-01-03:
I also have same problem.

WS2812B - 12 LEDs
GPIO 10, SPI enabled, all tests pass
at first wired directly and now via level shifter (also checked voltages using multimeter 3,3V vs 5V)
solid power supply (verified with multimeter - 3A on strip connectors)
3,3V and 5V connected to common ground for rpi, level shifter, led strip

there is no signal (verified with multimeter) between ground and GPIO10 (while on, at 3,3V and 5V lines)

```
[2023-01-03 09:43:22,632] INFO: Debug logging not enabled, if you are reporting issues please enable it under 'Features' in the settings page.
[2023-01-03 09:43:22,640] INFO: Starting main loop
[2023-01-03 09:48:42,892] INFO: Switching lights on
pi@octopi:~ $ tail -f /home/pi/.octoprint/logs/plugin_ws281x_led_status_debug.log
 | - | - | - enabled: True
 | - | - | - time: 750
 | - sacrifice_pixel: False
 | - at_command_reaction: True
 | - intercept_m150: True
 | - debug_logging: True
[2023-01-03 09:57:16,910] DEBUG: Returning to previous state: {'effect': 'Solid Color', 'color': '#0000ff', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 09:57:16,910] DEBUG: Changing effect to Solid Color
[2023-01-03 09:57:16,914] INFO: Starting main loop
[2023-01-03 09:57:16,916] DEBUG: New message: {'type': 'lights', 'action': 'on'}
[2023-01-03 09:58:22,377] DEBUG: New message: {'type': 'lights', 'action': 'off'}
[2023-01-03 09:58:22,379] INFO: Switching lights off
[2023-01-03 09:58:23,135] DEBUG: Changing effect to blank
[2023-01-03 09:58:23,136] DEBUG: Blanking LEDs
[2023-01-03 09:58:27,186] DEBUG: New message: {'type': 'lights', 'action': 'on'}
[2023-01-03 09:58:27,188] INFO: Switching lights on
[2023-01-03 09:58:27,191] DEBUG: Changing effect to Solid Color
[2023-01-03 09:58:34,208] DEBUG: New message: {'type': 'standard', 'effect': 'torch'}
[2023-01-03 09:58:34,210] DEBUG: Changing effect to torch
[2023-01-03 09:58:40,731] DEBUG: New message: {'type': 'standard', 'effect': 'torch'}
[2023-01-03 09:58:40,734] DEBUG: Changing effect to torch
[2023-01-03 09:58:51,417] DEBUG: New message: {'effect': 'Solid Color', 'color': '#000000', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 09:58:51,418] DEBUG: Changing effect to Solid Color
[2023-01-03 09:58:55,474] DEBUG: New message: {'effect': 'Solid Color', 'color': '#00ff00', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 09:58:55,476] DEBUG: Changing effect to Solid Color
[2023-01-03 09:59:00,505] DEBUG: New message: {'effect': 'Solid Color', 'color': '#ff0000', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 09:59:00,507] DEBUG: Changing effect to Solid Color
[2023-01-03 09:59:08,291] DEBUG: New message: {'effect': 'Solid Color', 'color': '#0000ff', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 09:59:08,293] DEBUG: Changing effect to Solid Color
[2023-01-03 09:59:10,971] DEBUG: New message: {'effect': 'Solid Color', 'color': '#ffffff', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 09:59:10,972] DEBUG: Changing effect to Solid Color
[2023-01-03 10:05:00,270] DEBUG: New message: {'effect': 'Solid Color', 'color': '#00ff00', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 10:05:00,271] DEBUG: Changing effect to Solid Color
[2023-01-03 10:12:32,332] DEBUG: New message: {'effect': 'Solid Color', 'color': '#0000ff', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 10:12:32,334] DEBUG: Changing effect to Solid Color
[2023-01-03 10:12:32,813] DEBUG: New message: {'effect': 'Solid Color', 'color': '#00ff00', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 10:12:32,815] DEBUG: Changing effect to Solid Color
[2023-01-03 10:12:44,117] DEBUG: New message: {'type': 'lights', 'action': 'off'}
[2023-01-03 10:12:44,120] INFO: Switching lights off
[2023-01-03 10:12:44,875] DEBUG: Changing effect to blank
[2023-01-03 10:12:44,876] DEBUG: Blanking LEDs
[2023-01-03 10:19:23,740] DEBUG: New message: KILL
[2023-01-03 10:19:23,741] DEBUG: Kill message received, shutting down...
[2023-01-03 10:19:23,742] DEBUG: Blanking LEDs
[2023-01-03 10:19:23,846] INFO: Effect runner shutdown. Bye!
[2023-01-03 10:19:24,580] DEBUG: Starting WS281x LED Status Effect runner
[2023-01-03 10:19:24,590] DEBUG: Current settings:
 STRIP SETTINGS
 | - count: 12
 | - pin: 10
 | - freq_hz: 800000
 | - dma: 10
 | - invert: False
 | - channel: 0
 | - reverse: False
[... 76 lines cut ...]
 FEATURES SETTINGS
 | - active_times
 | - | - enabled: False
 | - | - start: 09:00
 | - | - end: 21:00
 | - transitions
 | - | - fade
 | - | - | - enabled: True
 | - | - | - time: 750
 | - sacrifice_pixel: False
 | - at_command_reaction: True
 | - intercept_m150: True
 | - debug_logging: True
[2023-01-03 10:19:24,592] DEBUG: Returning to previous state: {'effect': 'Solid Color', 'color': '#00ff00', 'delay': 10, 'trigger': 'test', 'type': 'custom'}
[2023-01-03 10:19:24,594] DEBUG: Changing effect to Solid Color
[2023-01-03 10:19:24,595] DEBUG: Blanking LEDs
[2023-01-03 10:19:24,600] INFO: Starting main loop
[2023-01-03 10:19:24,602] DEBUG: New message: {'type': 'lights', 'action': 'off'}
[2023-01-03 10:20:01,681] DEBUG: New message: {'type': 'lights', 'action': 'on'}
[2023-01-03 10:20:01,683] INFO: Switching lights on
[2023-01-03 10:20:01,686] DEBUG: Changing effect to Solid Color
[2023-01-03 10:20:02,330] DEBUG: New message: {'type': 'lights', 'action': 'on'}
![Uploading Zrzut ekranu 2023-01-3 o 11.24.54.png…]()

```

imkebe, 2023-01-03:
Also tested with another RPI as a drop in replacement RPI2 -> RPI3B. Nothing changed

cp2004, 2023-01-03:
> there is no signal (verified with multimeter) between ground and GPIO10

I'm not sure if you will be able to pick up anything with a multimeter anyway - I haven't tried it, but my initial suspicion would be that the signal would be too quick for most multimeters to register anything. A scope may be better if you want to read the signals.

Can you share a picture of the wiring to show how you have it setup?

nnw522, 2023-01-03:
I have gone through all these issues to get this to work on my setup. I
found that the GPIO and physical Pins are different. I bought one of these
to help me with this issue

https://www.amazon.com/gp/product/B08RDYDG6X/ref=ppx_yo_dt_b_search_asin_title?ie=UTF8&psc=1

I just temporarily put this on to see which physical pin was actually
getting the signal. just go to the tests and keep sending the signal. you
should see one of the pins lighting up faintly.

nnw522, 2023-01-03:
> ^^
> I'm reffering to GPIO10 known as MSIO SPI. Also checked GPIO20 with SPI1
> enabled but there is an error then.
> I've checked my strips with dumb 5V power + GND + 3V DI PIN activation
> without RPI and they light up (without level shifter - 4,7V-5V vs 3,2V
> measured).
> So again I connected strip to RPI. Voltages/Power are OK however strip
> won't light up. Direct GPIO10 and via level shifter.
>
> Here is wiring with level shifter. Direct one is without it just
> GPIO10-DIN.
> [image: rpi-ws2812b_bb]
> <https://user-images.githubusercontent.com/4148157/210406166-438d7bd5-60db-46b7-a624-2980935ed797.png>

that level shifter did not work for me, i have an ic chip level shifter
that worked for me.  74AHCT125 is the part number I used. I might have some
spares somewhere when i get home tonight

nnw522, 2023-01-03:
i used the IC for Neo Pixel and work light control. it has 4 channels

nnw522, 2023-01-03:
sorry for the many replies, but also i used some capacitors on the power
rail to help with glitching.

imkebe, 2023-01-03:
I'll try with IC module.

djepsylon, 2023-01-04:
> @djepsylon If you still need help please do reply.

sorry for long response i was have a big trouble in my life but now it is solved (wow).
now for our oproblem, i think the problem come from the gpio of my pi i have buyed a new one and when i have it i testy if ws28xx works correctly

imkebe, 2023-01-07:
In my case I found that installing one of :
- rpi_ws281x
- adafruit-circuitpython-neopixel
- python3-rpi.gpio
solved problem.

I'm running OctoPring 1.8.6 @ OctoPi 0.18.0 and one of those packages were missing.

djepsylon, 2023-01-11:
with the new raspberry everything works good. but i've installed the packages in the @imkebe post. witout this, no access to gpio

cp2004, 2023-01-11:
Great news 👍

Thanks for updating the issue. I'll close this now, but feel free to ask any further questions.

closed 2023-01-11 by cp2004

imkebe, 2023-01-13:
@cp2004 consider adding information abour that packages to the troubleshooting documentation. I was adding debian (via apt install) packages not the python ones via pip.