cp2004/OctoPrint-WS281x_LED_Status #206: not getting signals from GPIO 10

orrinkuntz, 2023-01-02:
### Describe your problem

I have WS2812B lights, which I have hooked up according to the wiring diagram. However, there are no signals being received from GPIO 10. I have the settings configured correctly, and have passed all 5 OS configuration tests. Thank you for any help!

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

cp2004, 2023-01-02:
> Log files
> No response

😕 

Please, if you want help, fill in the template fully and upload the log files.

cp2004, 2023-01-22:
Since no log files were provided, I'm closing this issue. If you still need help setting the plugin up, then please respond to the issue.

closed 2023-01-22 by cp2004