cp2004/OctoPrint-WS281x_LED_Status #213: Multiple LED Strips

Eerier, 2023-02-24:
### Describe your problem

hey, is there a way to connect multiple LED Strips with individual settings, so i can have one configured as a flashlight for octolapse an the other one as a static progress bar. I could make both bars the same size and just wire them in parallel but then everything is flashing when octolapse takes a snapshot... 
Thank you for your help

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

cp2004, 2023-02-24:
It's not currently possible to connect multiple LED strips - the only way is to wire them together. I am aiming to add segment/multi-strip support to the plugin in the next release but I can't tell you when that will be. There's been a number of times this has been raised & I would find it quite useful myself, so it is already on the list. I'll close this issue but if you have any other questions please do ask.

closed 2023-02-24 by cp2004

Eerier, 2023-02-25:
Thank you very much!