cp2004/OctoPrint-WS281x_LED_Status #220: Flickering between correct colors and entirely white output on octoprint reinstall

JeremyLaurenson, 2023-09-26:
### Describe your problem

I reinstalled a new version of octoprint on an existing, working hardware and the WX plugin stopped working properly.

It starts out working and flashed fully white every so often and then will stick on full white.

I have a large external power supply running both as well as a shifter.

### Please confirm the following

- [X] I have read the Get Help & Troubleshooting guide

### Log files

[octoprint.log](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/12729349/octoprint.log)
[pluginlog.txt](https://github.com/cp2004/OctoPrint-WS281x_LED_Status/files/12729373/pluginlog.txt)

No more log messages after the last one.

JeremyLaurenson, 2023-09-26:
[2023-09-26 16:36:26,461] DEBUG: New message: {'type': 'progress', 'effect': 'progress_heatup', 'value': 99}
[2023-09-26 16:36:26,461] DEBUG: Changing effect to progress_heatup, 99%
[2023-09-26 16:36:27,466] DEBUG: New message: {'type': 'progress', 'effect': 'progress_heatup', 'value': 100}
[2023-09-26 16:36:27,466] DEBUG: Changing effect to progress_heatup, 100%
[2023-09-26 16:36:27,885] DEBUG: New message: {'type': 'progress', 'effect': 'progress_heatup', 'value': 100}
[2023-09-26 16:36:27,885] DEBUG: Changing effect to progress_heatup, 100%
[2023-09-26 16:36:28,462] DEBUG: New message: {'type': 'progress', 'effect': 'progress_heatup', 'value': 100}
[2023-09-26 16:36:28,462] DEBUG: Changing effect to progress_heatup, 100%
[2023-09-26 16:36:29,470] DEBUG: New message: {'type': 'standard', 'effect': 'failed'}
[2023-09-26 16:36:29,471] DEBUG: Changing effect to failed

cp2004, 2023-12-27:
So sorry for the delay in replying here - when you reinstall OctoPrint, you need to re-run the OS configuration checks. If the core frequency is not set correctly, then the timings for the LEDs are off and they tend to flash white.

From your log:
```
2023-09-26 16:16:48,449 - octoprint.plugins.ws281x_led_status - INFO - OS config test complete. Results:
| - adduser: True
| - spi_enabled: True
| - spi_buffer_increase: False !! Reason: failed
| - set_core_freq: True
| - set_core_freq_min: False !! Reason: failed
```

Once again, life was absolutely hectic the last few months and will continue to be, but I don't ever remember seeing this issue as I'm sure I would have replied! Sorry about that.

Please let me know if you still need help.