cp2004/OctoPrint-WS281x_LED_Status #221: [Request]: Custom color order for SK6812 compatible strips

pawelsky, 2024-03-22:
### What would you like implemented?

I recently came across a WS2814/SK6812 compatible strip with (unusual?) WRGB color order, which I unfortunately cannot configure properly to use with this great plugin. The closest I can get is SK6812_STRIP_BRGW but the Blue and White are swapped.

While it probably does not make sense to add each possible color order to the dropdown menu it would be good to have an option to define a custom color order.

The easiest (but probably not most user friendly) way to do it would be to do something similar to the underlying _rpi_ws281x_ library's SK6812_SHIFT_WMASK where use could define the custom shift mask in hex. More user friendly UI would of course be welcomed, but I can live with hexes hidden somewhere in the advanced section :)

### Why should this be implemented?

To provide compatibility with more LED strip types

cp2004, 2024-05-09:
If the underlying library supports it, then I am happy for it to be in the plugin. I don't currently have time to do this or the hardware to test it with (I don't actually have SK6812 strips) - but if someone contributed it I would happily review & merge.

This plugin is long overdue an update, but I cannot put a timeframe on when I might be able to sink some time into it for now.

Apologies for the *really* delayed answer here, I completely missed this issue and only came looking because of the latest one that was opened.

pawelsky, 2024-05-10:
To my understanding the underlying library does not support it, but I have a code modification that would enable that. I'll propose it to them and let you know if they decide to implement it.

cp2004, 2024-05-10:
Great, thanks. Keep me updated, and I'll be happy to weigh in with opinions/support if it helps push it along.

pawelsky, 2024-05-12:
Raised an issue here: https://github.com/jgarff/rpi_ws281x/issues/540